hanruihua/NeuPAN #3: index error when I run "python run_exp.py -e pf_obs -d diff "

TravisHC, 2025-04-01:
this is the log when error occurs
Traceback (most recent call last):
  File "/home/travis/code/NeuPAN/example/run_exp.py", line 76, in <module>
    main(env_path_file, planner_path_file, args.save_animation, ani_name, args.full, args.no_display, args.point_vel)
  File "/home/travis/code/NeuPAN/example/run_exp.py", line 42, in main
    env.draw_points(neupan_planner.dune_points, s=25, c="g", refresh=True)
  File "/home/travis/code/ir-sim/irsim/env/env_base.py", line 208, in draw_points
    self._env_plot.draw_points(points, s, c, refresh, **kwargs)
  File "/home/travis/code/ir-sim/irsim/env/env_plot.py", line 224, in draw_points
    y_coordinates = [point[1] for point in point_list]
                     ~~~~~^^^
IndexError: index 1 is out of bounds for axis 0 with size 1
INFO: Simulated Environment End with sim time elapsed: 4.8 seconds

maybe the bug is caused by empty point in non-empty point_list, i fix it as follows in env_plot.py

`    def draw_points(self, point_list, s=10, c="m", refresh=True, **kwargs):
        """
        Draw points on the plot.

        Args:
            point_list (list): List of points, each point as [x, y].
            s (int): Size of the points.
            c (str): Color of the points.
            refresh (bool): Whether to refresh the plot.
            kwargs: Additional plotting options.
        """
        if point_list is not None:

            if (len(p) == 0 for p in point_list):
                 # print("No points to draw")
                 return

            x_coordinates = [point[0] for point in point_list]
            y_coordinates = [point[1] for point in point_list]

            points = self.ax.scatter(x_coordinates, y_coordinates, s, c, **kwargs)

            if refresh:
                self.dyna_point_list.append(points)`

hanruihua, 2025-04-01:
Hi,

This error may be casued by the old version of ir-sim. 

You can run `pip install ir-sim -U` to install the newest version ir-sim 2.4.0 and run these examples again.

TravisHC, 2025-04-01:
cool, it works. 谢谢佬

closed 2025-04-01 by TravisHC