hanruihua/NeuPAN #6: Help！robot cannot pass through the narrow channel

caiziheng, 2025-04-21:
Hello, I'm very interested in your achievements. I deployed your code at the first time, but there is one problem that I can't solve. I hope to get your help. No matter how I change q_s, p_u, eta, d_max, d_min, bk, ro_obs, robot cannot pass through the narrow channel. The following is the specific yaml configuration
I adjusted the size of the robot to the value I wanted: robot:
- kinematics: {name: 'acker'}
shape: {name: 'rectangle', length: 1.9, width: 1.1, wheelbase: 1.12}
state: [-5, 20, 0, 0]
goal: [45, 20, 0]
vel_min: [-4, -1.5]
vel_max: [4, 1.5]
goal_threshold: 0.3
plot:
show_trail: True
show_goal: True
show_trajectory: True

sensors:
- type: 'lidar2d'
range_min: 0
range_max: 20
angle_range: 3.1415926
number: 200
noise: False

Environmental obstacles:
- number: 6
distribution: {name: 'manual'}
shape:
- {name: 'rectangle', length: 70, width: 2}
- {name: 'rectangle', length: 70, width: 2}
- {name: 'rectangle', length: 5, width: 2}
- {name: 'rectangle', length: 10, width: 2}
- {name: 'rectangle', length: 6, width: 2}
- {name: 'rectangle', length: 5, width: 2}

State: [[30, 25, 0], [15, 30, 0], [5, 18.5, 1.57], [23, 23.4, 1.57], [36, 17, 2.1], [50, 22, 4.3]]

receding: 20
step_time: 0.2
ref_speed: 4
device: 'cpu'
time_print: True
collision_threshold: 0.001

robot:
kinematics: 'acker'
max_speed: [4, 1.0]
max_acce: [4, 0.5]
length: 1.9
width: 1.1
wheelbase: 1.12

ipath:
waypoints:  [[0, 20, 0], [15, 20, 0], [45, 20, 0]]
curve_style: 'dubins'   # 'dubins' or reeds
min_radius: 3.0
loop: False
arrive_threshold: 0.1
close_threshold: 0.1
ind_range: 10
arrive_index_threshold: 1

pan:
iter_num: 2
dune_max_num: 200
nrmp_max_num: 30
dune_checkpoint: 'example/model/acker_robot_default/model_5000.pth'
iter_threshold: 0.1

Thanks！

caiziheng, 2025-04-21:
[image]

Even unimaginable trajectory lines emerged

caiziheng, 2025-04-21:
At the beginning, I manually adjusted q_s, p_u, eta, d_max, d_min, bk, and ro_obs. Later, I directly traversed in the loop, but still couldn't obtain the result.

hanruihua, 2025-04-21:
Hi, thanks for your interest.

If you adjusted the size of the robot, the dune model should be retrained, otherwise the minimum distance calculation will have wrong results. 

You can refer to the example dune_train: https://github.com/hanruihua/NeuPAN/tree/main/example/dune_train, to train the model for your specific robot size. After training, make sure you have selected this model in the YAML file.

caiziheng, 2025-04-21:
Cool，I will conduct an in-depth study on how to train.
thanks again!

caiziheng, 2025-04-22:
hi, I have retrained the model according to my own size, but this problem still occurs, that is, the red trajectory will not appear in the channel below. I adjusted those six parameters, but it not works. May I ask how I should solve this situation? Thank you!

[image]

[image]

caiziheng, 2025-04-22:
I modified the information of the obstacles, but the red trajectory is unreasonable. How should I solve this problem？

[image]

[image]

[image]

At this point, tell me "NeuPAN stop because of minimum distance", and then exit.

hanruihua, 2025-04-23:
Hi,

The receding step is too large. A long receding step can increase the difficulty of finding a solution and require more iterations to converge. It is recommended to use a shorter receding interval, for example, 5 or 10.

Additionally, could you please provide the full yaml files (env and planner) so that I can know all the parameters setting exactly to identify the cause of the error.

caiziheng, 2025-04-24:
hi,
When I reduce  receding step, it becomes even more difficult to find an effective trajectory. Here is all yaml files.
env.yaml
world:
  height: 23
  width: 70
  step_time: 0.1
  sample_time: 0.1
  offset: [-3, 10]
  collision_mode: 'stop'  # 'stop', 'unobstructed', 'reactive'
  control_mode: 'auto'  # 'keyboard', 'auto'

robot:
  - kinematics: {name: 'acker'} 
    shape: {name: 'rectangle', length: 1.8, width: 0.9, wheelbase: 1.12}
    state: [-5, 20, 0, 0] 
    goal: [45, 20, 0] 
    vel_min: [-8, -1.5]
    vel_max: [8, 1.5]
    goal_threshold: 0.3
    plot:
      show_trail: True
      show_goal: True
      show_trajectory: True

    sensors: 
      - type: 'lidar2d'
        range_min: 0
        range_max: 20
        angle_range: 3.1415926
        number: 200
        noise: False
      
      
obstacle:
  - number: 6
    distribution: {name: 'manual'}
    shape:
      - {name: 'rectangle', length: 70, width: 2} 
      - {name: 'rectangle', length: 70, width: 2}
      - {name: 'rectangle', length: 5, width: 2}
      - {name: 'rectangle', length: 6, width: 2}
      - {name: 'rectangle', length: 6, width: 2}
      - {name: 'rectangle', length: 5, width: 2}

    state: [[30, 25, 0], [30, 15, 0], [5, 18.5, 1.57], [23, 20, 1.57], [36, 17, 2.1], [50, 22, 4.3] ]

planner.yaml
# mpc
receding: 35
step_time: 0.1
ref_speed: 4
device: 'cpu'
time_print: True
collision_threshold: 0.001

robot:
  kinematics: 'acker'
  max_speed: [4, 3]
  max_acce: [4, 1.5]
  length: 1.8
  width: 0.9
  wheelbase: 1.12

ipath:
  waypoints:  [[-5, 20, 0], [45, 20, 0]]
  curve_style: 'dubins'   # 'dubins' or reeds
  min_radius: 3.0
  loop: False
  arrive_threshold: 0.1
  close_threshold: 0.1
  ind_range: 10
  arrive_index_threshold: 1

pan:
  iter_num: 3
  dune_max_num: 200
  nrmp_max_num: 20
  # dune_checkpoint: 'example/model/acker_robot_default/model_5000.pth'
  dune_checkpoint: "/home/stephen/Workplace/RL_Methods/NeuPAN/models/model_5000.pth"
  iter_threshold: 0.1

adjust:
  q_s: 0.6   #1
  p_u: 0.8     #1
  eta: 15     #10
  d_max: 1    #1
  d_min: 0.1  #0.1
  bk: 0.1     #0.1
  ro_obs: 400 #400

hanruihua, 2025-04-25:
Hi,

First, the maximum speed specified in planner.yaml should not exceed the value in the env.yaml file.

Second, the environment you provided is quite challenging, making it difficult for a local planner to handle the situation. Therefore, you can modify the initial path by adding some waypoints; for example, add a waypoint before the narrow channel to guide the robot in advance.

caiziheng, 2025-04-27:
Hi,
Thanks for your advice, I will try later.

Thanks again!

caiziheng, 2025-04-28:
Hi,
In the new test, I discovered a new problem, that is, the generated trajectory lines might pass through obstacles. This is contrary to the real situation and is also undesirable. I made the following modifications to the minimum boundary constraint, and this situation will no longer occur.
`delete   constraints += [cp.min(self.indep_dis) >= self.para_d_min]`
and 
`add constraints += [ self.indep_dis >= self.para_d_min ]`
Hope to make your code more perfect.

caiziheng, 2025-04-28:
my pleasure