hanruihua/NeuPAN #7: Simulation degrades after changing ref_speed from 4 to 1

tanz0710, 2025-04-24:
Hi team,

Thanks for your great work on this project.

I’ve encountered an issue when adjusting the simulation settings. Specifically, I changed the ref_speed parameter from 4 to 1, and noticed that the robot’s obstacle avoidance performance significantly deteriorated in the simulation.

I checked the README and documentation, but didn’t find any training parameters that seemed directly tied to ref_speed. Could you please advise on:

Whether the policy is sensitive to the ref_speed parameter during inference?

Should the policy be retrained or fine-tuned if I want to run with a lower ref_speed?

Looking forward to your suggestions. Thanks again!

tanz0710, 2025-04-24:
Here is the command I use to run the experiment:

python example/run_exp.py -e polygon_robot -d diff

hanruihua, 2025-04-25:
Hi,

The `ref_speed` parameter is associated with two main features of this planner: prediction and the reference speed cost.

The current receding length is determined by:  
`ref_speed` * `step_time` * `receding`.  
Consequently, a lower `ref_speed` shortens the receding length and negatively affects prediction performance, meaning that a large size robot might not avoid obstacles in time. In such cases, the `step_time` or `receding` value can be increased.

The `ref_speed` parameter is used to calculate the speed cost to ensure the robot runs at a specific speed. A lower `ref_speed` could reduce this cost, making the robot more conservative. In this case, the weight parameter `p_u` for this cost can be increased, and the `max_sd` can be decreased.

Besides, the policy does not need to be retrained since the DUNE model is only related to the robot's geometry size. Fine-tuning the adjustment parameters is feasible if the environment is particularly complex; otherwise, you can simply manually adjust the parameters.

I changed `step_time` to 0.2, `p_u` to 2.0, and `d_max` to 0.5. The results are shown below:

[image]

tanz0710, 2025-04-27:
Thank you very much for your excellent open-source project. It's truly impressive and well-designed.

Following your explanation, I adjusted the p_u parameter as you suggested, and it effectively solved the issue I was encountering.
I have successfully applied your project on the Unitree Go2 robot and achieved obstacle avoidance.

Thanks again for your contribution!

closed 2025-04-27 by tanz0710