hanruihua/NeuPAN #8: Questions about 3D Lidar

caiziheng, 2025-04-28:
Hello, it's me again.

I now want to deploy the adapted code on a real vehicle, but the Lidar on the vehicle is 3D. Although I can detect the Lidar points and turn them into 2D, why not directly use the 3D Lidar information?

I reviewed your paper again, but except that the time complexity was slightly mentioned, there is no other reason to tell me why. So is the only reason that the number of points on the 3D radar is too large, resulting in a slower reasoning speed?

There is one problem that cannot be solved when using 2D Lidar. When the height of the obstacle needs to be taken into consideration. For example, when a vehicle needs to pass through a passage where there is an obstacle hanging down above, it is necessary to know whether this obstacle can collide with the top part of the vehicle. One mitigation method I can think of is to use 3D Lidar, but filter the Lidar points according to the height of the vehicle and then turn it into 2D. Is that so?

Thank you again for your patience and time.

hanruihua, 2025-04-29:
Hi,

In theory, 3D space should work well. However, this project only considers robot navigation in 2D space, both the kinematic constraints and the DUNE model are set for 2D. Therefore, the 3D points are mapped to 2D space by filtering out the z coordinate.

The effect and performance are the same as directly using the 3D lidar information, since all the 3D points within the specified z-range are considered (depending on the vehicle's height), as you pointed out.

For your case, we also conducted experiments on a real vehicle equipped with a 3D lidar. We used the [pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan) package to convert the 3D point cloud to 2D scan data. The parameters `min_height` and `max_height` can be adjusted based on the vehicle’s height to handle the obstacle points hanging above.

caiziheng, 2025-04-29:
Thank you again! I will try later

closed 2025-04-29 by caiziheng