hanruihua/NeuPAN #9: nrmp使用排序后的少量点似乎容易在“路径规划”层面犯错？

xuwenjie401, 2025-05-08:
首先，很感谢能开源这么棒的工作
以下是我试用过程中发现的问题

https://github.com/user-attachments/assets/6c766bfd-3272-4e03-aaf5-2737dba385b6

https://github.com/user-attachments/assets/9a429523-1ee4-45e1-b8ad-9e01f7e6ec8a

首先，按照README自己训了个DUNE，发现corridor这里，很容易在避障时走入死角或者一直转圈，（去掉两边的墙壁的话成功率就很高）这是因为nrmp只使用按距离排序的部分点导致的吗（刚好这个场景的三边距离卡到了某个阈值？）？但nrmp把输入点数调高好像很容易求解崩溃，这是什么原因呢？

这种问题期望的解决方法是什么呢？修改waypoints引导吗 —— 那对动态障碍物是不是会存在类似的问题？

xuwenjie401, 2025-05-10:
OK，非常感谢解答。
我不是做规控的，所以还有一个问题是，相比于
“hybrid-A*+多边形碰撞检测做路径规划，轨迹跟踪用无碰撞约束的较简单的MPC“
neupan这种用极简单的路径规划 + 考虑碰撞约束的轨迹跟踪的优势，是对高动态环境的求解速度吗（或者是端到端路线的探索意义？）

hanruihua, 2025-05-11:
主要优势在于:

1.  控制的精度更高： 因为像传统的框架需要中间模块做目标的检测或者建立地图，都会存在误差传递，而端到端的架构就避免了这些误差，从而在更复杂更极端狭窄的环境下效果更好。

2. 架构更灵活： neupan的架构是基于mpc的优化架构，因此可以根据不同的环境和任务添加或者修改不同的约束与代价函数，易于迁移到于不同的平台，例如差速轮和阿克曼模式的切换就是修改对应的动力学约束。而传统的架构，例如hybrid-A 是专用于阿克曼的，切换不同的平台要修改非常多的模块。

3. 更易部署：传统的框架模块多，参数多，虽然容易调试，但是每个模块都需要调参编译配环境，而neupan只需要传感器端发送数据和控制端接收命令即可，更易在不同的平台快速部署。相比于其他的learning的端到端，neupan的训练也很简单，而且是可以落地的。

至于高动态环境的求解速度，因为是基于优化的框架，所以对于CPU算力有一定的要求，算力越高，求解速度越快，在高动态环境表现就越好了。

xuwenjie401, 2025-05-11:
好的，再次感谢解答

closed 2025-05-11 by xuwenjie401