hanruihua/NeuPAN #10: 运行example报错

XinJingHao, 2025-05-19:
很好的工作！ 不过我在运行example时遇到点问题，求助！

我将代码clone到本地，并安装好依赖后，执行：
```
cd example 
python run_exp.py -e non_obs -d acker

报错：ModuleNotFoundError: No module named 'neupan'
```

我尝试将neupan文件夹放入example目录下，再运行：
```
cd example 
python run_exp.py -e non_obs -d acker

提示：No train kwargs provided. Default value will be used.
Do not find the DUNE model; Do you want to train the model now, input Y or N:
```
很奇怪，明明在```example /model```下有看到保存的```.pth```文件呀。

请问我哪一步出问题啦？

hanruihua, 2025-05-19:
你好，

代码clone到本地之后，需要在NeuPAN文件夹内运行

`pip install -e .`

将 neupan 包安装到本地之后再运行。你的报错显示 neupan 没有安装成功。

XinJingHao, 2025-05-19:
原来如此！  成功运行了，非常感谢！
确实是```pip install -e .```这一步偷了懒造成的- -。。

closed 2025-05-19 by XinJingHao